TheThingsIndustries/generic-node-se #106: Support LoRaWAN OTAA and ABP Join best practices

elsalahy, 2021-01-27:
**Summary:** 

Support LoRaWAN OTAA and ABP Join best practices.
See [article](https://www.thethingsnetwork.org/docs/devices/bestpractices.html) for more information.

CC: @htdvisser 

**Why do we need this?**

To allow for scalable deployments.

**What is already there? What do you see now?**

Demo applications `basic_lorawan` and `freertos_lorawan` that doesn't follow the best practices.

**What is missing? What do you want to see?**

Demo applications following best practices.

**How do you propose to implement this?**

- Collect all practical best practices under the current umbrella issue
- Modify the LoRaWAN basic apps to follow the best practices.

**Environment:**

Baremetal and FreeRTOS.

**Acceptance Criteria:**

- Reference apps maintain their low power performance.

**What can you do yourself and what do you need help with?**

All

htdvisser, 2021-01-27:
Related: 

https://www.thethingsnetwork.org/forum/t/does-a-node-ever-need-to-rejoin-after-otaa/2442/10?u=htdvisser
https://www.thethingsnetwork.org/forum/t/does-a-node-ever-need-to-rejoin-after-otaa/2442/12?u=htdvisser

elsalahy, 2021-07-05:
@marnix, can you please break down this issue into smaller manageable issues so that we can start implementing the best practices

marnix, 2021-07-05:
That comment from @elsalahy is presumably not for me but for @marnixcro. :-)

elsalahy, 2021-07-06:
Indeed @marnix, it was intended for @marnixcro 😅