TheThingsIndustries/generic-node-se #109: Reduce enclosure size by using AAA and/or flat li-poly batt.

ehogeweg, 2021-01-27:
**Summary:** 
The node is spectacular but seems a tad bulky. Changing the batteries would allow for a smaller enclosure. Additional... it would be nice (...) if the battery can be charged with a little solar panel on the cover.

**How do you propose to implement this?**
It seems easy enough to define a different bottom, maybe even include both bottoms in the package. Keeping two different models on stock seems overkill.
...

**Environment:**

...

**Acceptance Criteria:**

...

Unable to help with the enclose but more than happy to help implementing the solar charging part.

azerimaker, 2021-01-27:
Thanks for your comment, @ehogeweg . That's certainly doable. In fact, that was the reason why we went with JST-PH connector, so you can use flat LiPo batteries. 
The reason why we went with 2xAA instead of 2xAAA was mainly due to the provided extra capacity for 2xAA ~9Wh which was almost 60%  more than the 2xAAA based solution. 
Energy harvesting can be implemented on a small add-on board which sits between a rechargeable battery and the main PCB.

azerimaker, 2021-01-27:
@ehogeweg please see this discussion for more info: https://github.com/TheThingsIndustries/generic-node-se/issues/8#issuecomment-624648301

ehogeweg, 2021-01-27:
That is an excellent thread, thanks for sharing. Too bad it is close ;-)

Keep up the good work guys. The generic node appears to be an awesome product. Can’t wait to get my hands on it.

Didn’t hear about it until today so I have a bit of catching up to do.

Cheers,

Erwin

ehogeweg, 2021-01-27:
Thanks Orkhan,

BTW… I seems like most of your team is in EUR. We are located in the US (South Florida) so if you need some beta testers here, we are more than happy to help.

Cheers,

Erwin

azerimaker, 2021-01-27:
Thanks Erwin (@ehogeweg ). 

Thanks, in that and other threads (mostly closed) you can have a general overview as we progressed in our design decisions. 

We'd love to have beta testers in USA. We have designed the device in such a way what it would support dual band **(868/915 MHz, +14 dBm/ +22dBm )** while sticking to one BOM and antenna matching network. This of course results in overall reduced efficiency, but that's a compromise for now. Later we might have EU and US only versions. 

Please join our [early access program](https://www.genericnode.com/) and we will send you more details.

closed 2021-02-10 by elsalahy