TheThingsIndustries/generic-node-se #113: lorawan freefall app

elsalahy, 2021-02-01:
**Summary:** 

We need an application that configures the accelerometer interrupt pin to report free-fall over LoRaWAN.

**Why do we need this?**

To have a sample application demonstrating the GNSE capability of detecting free-fall.

**What is already there? What do you see now?**

basic_lorawan application

**What is missing? What do you want to see?**

A freefall demonstration app.

**How do you propose to implement this?**

- The applications connects OTAA over LoRaWAN
- The application sends an uplink when the device detects a freefall.
- If the device can't send a message it accumulates the freefall events and send them in the next transmission. 
- Any parameters that can be configured, should be in app_conf.

**Environment:**

Baremetal

**Acceptance Criteria:**

- App compiles with no warnings
- App functions as expected

**What can you do yourself and what do you need help with?**

Marnix will handle this application and ask for direction when needed.

elsalahy, 2021-02-02:
Here is a similar [application](https://lora-developers.semtech.com/resources/lorawan-academy/) we developed using another platform.

elsalahy, 2021-02-02:
@marnixcrouzen please feel free to discuss the requirements before jumping into implementation.

marnixcrouzen, 2021-02-11:
I have a couple of questions about this feature

1. Wat message should appear in the payload?
2. Has there already been an implementation like the third item on the proposal list? I first thought that the button events from basic_lorawan would have this feature but it only notifies the user and never sends the message. It should be doable by setting a timer on failed attempts, but maybe I'm missing a more elegant build-in solution.

elsalahy, 2021-02-12:
@marnixcrouzen 

1. You have flexibility with the message structure but it keep in mind it should be as configurable as possible.
2.  Yeah you can use a configurable transmission timer using the sequencer.

PR #120 (2021-03-04): Feature/freefall app
**Summary:**

A basic implementation of the Freefall App. The node connects over OTAA and waits for free fall events. If they occur, then the amount of free falls is added to a payload and transmitted. Closes #113.

**Changes:**

- A new folder in the app dir (with the basic_lorawan as base code)
- The changes required to setup the freefall app (mainly in lora_app.c, freefall.c, app_conf.h)
- Initialisation (and detinitialisation) functions for the accelerometer pin in GNSE_bsp_gpio.c
- Changed Cmakelists.txt to add freefall_lorawan as app

**Notes for Reviewers:**

The de [... cut]
closed 2021-03-04 by marnixcrouzen