TheThingsIndustries/generic-node-se #114: [HW] Boot mode pins capacitance too high

marnixcrouzen, 2021-02-04:
**Summary:** 

The boot pin connection seems to have a capacitance that's too high. My node jumped into boot mode without connecting the boot pins together. The device was power cycled and reset multiple times in between. Discharging power supply by shorting it was also not successful. The issue was fixed after turning the device off, shorting the pins temporarily, and turning it on again.

**Steps to Reproduce:** 

1. Shorting the boot pins with a header
2. Turning on the device for some time
3. Disconnecting the header
4. Resetting
4. Try to connect to the device via UART (if connected, the boot mode is running)

**What do you see now?**

...

**What do you want to see instead?**

...

**How do you propose to implement this?**

...

**Environment:**

Bare metal

**What can you do yourself and what do you need help with?**

...

elsalahy, 2021-02-05:
This is a good first issue, from the title to the Summary, will done @marnixcrouzen 👏 !

azerimaker, 2021-02-11:
@marnixcrouzen thank you for the issue. Could you please verify if there is a short on BOOT0 pins under the PCB?
I typically measure >50KOhm resistance between the BOOOT0 and VDDRF, and couldn't reproduce it. 
[image]
OT0 pins.

marnixcrouzen, 2021-02-11:
@azerimaker the node is not shorted on the boot0 pins. When I resolved the issue I was actually trying to find short there but that didn't seem to be the case then. After testing it and knowing there was no short, it unexpectedly worked after turning it on. I thought that shorting it during the process of testing with my probes might have discharged it.

I tried reproducing it myself today. I wasn't able to get it into boot mode without shorting the pins. It is either harder to reproduce than I thought or something else caused the issue.

An alternative theory that I have is that the short was caused by some smudge between the pads. Maybe trying to touch the pads with my probes made me accidentally wipe the smudge off, clearing the short before determining that there was any. The pads are extremely close together, but I wouldn't know how it suddenly could be shorted after a full day of it working and even working the morning of the incident.

azerimaker, 2021-03-13:
Since this issue is not present in other nodes I'm closing it for now. We can reopen it we observe anything again.

closed 2021-03-13 by azerimaker