TheThingsIndustries/generic-node-se #127: Incorrect PCB Pin description

elsalahy, 2021-02-22:
**Summary:** 

The VCC pin on the front is wrongly described on the back as GND
and The GND pin on the front is wrongly described in the back as VCC

**Steps to Reproduce:** 

Check any PCB front and back.
[image]

[image]

**What do you see now?**

Incorrect pin description.

**What do you want to see instead?**

Correct pin description.

**How do you propose to implement this?**

Fix in next production phase.

**Environment:**

Hardware

**What can you do yourself and what do you need help with?**

@azerimaker can easily fix this.

azerimaker, 2021-02-24:
Good catch, @elsalahy. Noted down and will fix it in the final rev.

azerimaker, 2021-03-13:
This has to be addressed in documentation. I'll look into that. 
Closing the issue for now.

elsalahy, 2021-09-24:
I see from the new HW design that this issue is solved.

[image]
[image]

lnlp, 2021-11-10:
@elsalahy 

> I see from the new HW design that this issue is solved.

On above pictures with new HW design the silkscreen on PCB top side still lists "v1.1".

New hardware designs should have new versions of course but above picture does not confirm that.

It appears that either the new PCB has a bug in the silk screen or above picture is not the new HW design.
Which of these assumptions is correct?

elsalahy, 2021-11-24:
@lnlp I think @azerimaker increased the version to v1.1 b?

lnlp, 2021-11-24:
Maybe, but that is not what the version on the PCB silkscreen says.

azerimaker, 2021-12-13:
Hi @lnlp. 
Indeed the PCB silkscreen has not been updated to indicate b version, as we did not see a reason to do so, because there was basically no hardware or layout changes. It was simply a logo and silkscreen update.