TheThingsIndustries/generic-node-se #128: Improve antenna performance

elsalahy, 2021-02-25:
**Summary:** 

The node RSSI is in the range of -80 to -90

**Steps to Reproduce:** 

1. Flash the node with `basic_lorawan` app
2. Record gateway, application logs

**What do you see now?**

Bad RSSI

Attached traced of SF12 and SF08.
The node is used without enclosure

**What do you want to see instead?**

Good RSSI in the range of -40 to -60

**How do you propose to implement this?**

Fix Antenna placement and tune accordingly.

**Environment:**

V1.1 PCB (second iteration, 100 patch)

**What can you do yourself and what do you need help with?**

I can provide assistance with Firmware configurations and testing.
@azerimaker needs to analyze other nodes and confirm if the issue is present and decide on a plan to fix this.

elsalahy, 2021-02-26:
@marnixcrouzen can you please report the RSSI values of your node?

marnixcrouzen, 2021-02-26:
> @marnixcrouzen can you please report the RSSI values of your node?

Did two readings, no casing, indoors
Reading 1:
  "time": "2021-02-26T12:32:45.181342063Z",
  "data_rate": "SF12BW125",
  "gateway 1":
      "rssi": -118,
  "gateway 2":
      "rssi": -109,
      "snr": 4.2,
Reading 2:
  "time": "2021-02-26T12:34:22.413363371Z",
  "data_rate": "SF10BW125",
  "gateway 1": 
      "rssi": -114,
      "snr": -0.8,
   "gateway 2":
      "rssi": -113,
      "snr": -0.2,

elsalahy, 2021-02-26:
@marnixcrouzen thanks for confirming the issue.

elsalahy, 2021-04-06:
@azerimaker please comment and update this issue, there seems to be a concern from the community as well observed in #163 

We need to update this in order to respond to community questions.

elsalahy, 2021-06-29:
@azerimaker can you please update this issue with latest performance gains communicated by Ignion?

azerimaker, 2021-07-01:
The design went through "Antenna Matching Network Tuning" process and we're happy to announce that it's efficiency have been improved quite significantly. Antenna has been tuned for various use cases and frequency ranges. Respective efficiency gains are shown below. 
[image]

azerimaker, 2021-07-01:
Closing this issue for now, but will repoen it once we have the second batch of nodes.

closed 2021-07-01 by azerimaker