TheThingsIndustries/generic-node-se #129: CMake configuration - setting CMAKE_TOOLCHAIN_FILE

descartes, 2021-02-25:
**Summary:** 
Fresh git pull or download of release isn't ready to process the root CMakeLists.txt 

**Steps to Reproduce:** 
New window / workspace. Open folder. CMake reports error with no CMAKE_TOOLCHAIN_FILE defined.

**What do you want to see instead?**
Add set(CMAKE_TOOLCHAIN_FILE "cross.cmake") if appropriate

**How do you propose to implement this?**
I don't, I hack CMake and I don't have a good enough overview of the project structure

**Environment:**
Using latest VS Code 1.53.2 on macOS 10.14.6

**What can you do yourself and what do you need help with?**
Carry on providing smoke tests until I understand enough to be dangerous

elsalahy, 2021-02-26:
Thanks @descartes 

I fail to reproduce the issue, even with release files.

If you run the commands as mentioned in the [documentation](https://thethingsindustries.github.io/generic-node-docs/getting-started/se-sw/env-setup/cmd/) do you face an issue.

Because the command below sets the `CMAKE_TOOLCHAIN_FILE`
`cmake ../.. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=../../cross.cmake -DTARGET_APP=basic`

descartes, 2021-02-28:
VSCode?? As per my report??

The Docker option is 'interesting'.

I hold my hand up, I didn't read the documentation, I saw the CMake files and being up to my ears in Raspberry Pi Pico files and VS Code fully configured, just used that.

Perhaps detailing your preferences for VS Code would be useful. And doing a quick survey of developers preferences on IDE / building so you can head off issues at the path. I suspect many GenericCode people will be moving over from experience with ANO development environment - just pray it's not Arduino.

elsalahy, 2021-03-01:
@descartes
I think you are aware CMake can be used within VSCode.

I agree something like this [VSCode documentation](https://thethingsindustries.github.io/generic-node-docs/getting-started/se-sw/env-setup/vscode/) can be more detailed.

If you want a more detailed documentation on something specific related to your setup, I would appreciate it if you open a descriptive issue [here](https://github.com/TheThingsIndustries/generic-node-docs)

Let me know if there is still a technical issue with the CMAKE after you set the flag.

HeadBoffin, 2021-03-01:
I'll figure out the settings based on the command line docs and either raise an issue or a pull request.

This can be closed if you wish

elsalahy, 2021-03-01:
@descartes or @HeadBoffin  feel free to re-open this if you still face an issue.

closed 2021-03-01 by elsalahy