TheThingsIndustries/generic-node-se #130: CMake configuration - best way to select build target

descartes, 2021-02-25:
**Summary:** 
I can see the use of TARGET_APP and I can see the validation of correct target and I can set TARGET_APP to the target I want but it is not clear what the most approprate way of selecting the target.

**Steps to Reproduce:** 
Once you've got the basic target running, figure out how to choose a different target

**What do you want to see instead?**
Something in the README.md that says what the most appropriate way of doing this is.

**How do you propose to implement this?**
I don't, I want to know how you do it

**Environment:**
Using latest VS Code 1.53.2 on macOS 10.14.6

**What can you do yourself and what do you need help with?**
Carry of smoke testing

elsalahy, 2021-02-26:
Thanks @descartes 

We mention this briefly in the [documentation](https://thethingsindustries.github.io/generic-node-docs/getting-started/se-sw/env-setup/cmd/) towards the end.

```
You can also configure the following build fields in the cmake command:

TARGET_APP to build a different application (try -DTARGET_APP=basic_lorawan)
DCMAKE_BUILD_TYPE to Release instead of Debug to build a smaller application for release purposes
```

Do you feel this is not sufficient information to build other [applications](https://github.com/TheThingsIndustries/generic-node-se/tree/develop/Software/app/README.md)?

descartes, 2021-02-28:
It's under a heading of Windows - so deeply unlikely I'd ever be going there - in fact having followed that link I'm off for a shower. Perhaps a CMake settings / config section?

HeadBoffin, 2021-03-01:
qf #129, close and I'll see what I can achieve

elsalahy, 2021-03-01:
I agree this can be improved
I opened  an issue for tracking
https://github.com/TheThingsIndustries/generic-node-docs/issues/16

elsalahy, 2021-03-01:
@descartes or @HeadBoffin 
Very similar to LMN [vscode setting](https://github.com/Lora-net/LoRaMac-node/blob/master/.vscode/settings.json), you can set the following configurations.
```
"CMAKE_BUILD_TYPE":"Debug",
"CMAKE_TOOLCHAIN_FILE"="../../cross.cmake", (or absolute path to cross.cmake)
"TARGET_APP"="basic"
```

closed 2021-03-01 by elsalahy