TheThingsIndustries/generic-node-se #131: Beta 0.2.0 won't compile - missing references

descartes, 2021-02-25:
**Summary:** 
```
[build] [ 67%] Building C object app/basic/CMakeFiles/main.elf.dir/basic_sensor.c.obj
[build] In file included from generic-node-se-0.2.0-beta/Software/app/basic/basic_se_flash.c:24:
[build] generic-node-se-0.2.0-beta/Software/lib/ATECC608A-TNGLORA/tnglora_read_helper.h:26:10: fatal error: cryptoauthlib.h: No such file or directory
[build]    26 | #include "cryptoauthlib.h"
[build]       |          ^~~~~~~~~~~~~~~~~
[build] compilation terminated.
[build] make[2]: *** [app/basic/CMakeFiles/main.elf.dir/basic_se_flash.c.obj] Error 1
[build] make[2]: *** Waiting for unfinished jobs....
[build] [ 68%] Building C object app/basic/CMakeFiles/main.elf.dir/__/__/lib/ATECC608A-TNGLORA/hal_LoraMacNode.c.obj
[build] generic-node-se-0.2.0-beta/Software/lib/ATECC608A-TNGLORA/hal_LoraMacNode.c:26:10: fatal error: atca_hal.h: No such file or directory
[build]    26 | #include "atca_hal.h"
[build]       |          ^~~~~~~~~~~~
[build] compilation terminated.
```

**Steps to Reproduce:** 
Download release, edit CMakeLists.txt to point at cross.cmake, set location of compiler in cross.cmake, build ...

**How do you propose to implement this?**
This is more of an FYI as I have a `git clone` that is working - but you may want to remove the release ...

**Environment:**
VSCode on macOS, CMake 3.18.1 - tried with CLion which experiences the same issue. CLion has it's own internal copy of CMake, 3.19.2. CLion, like VS Code, can compile the `git clone` version.

**What can you do yourself and what do you need help with?**
Carry on cheering from the sidelines?

elsalahy, 2021-02-26:
Thanks @descartes 

We depend on two submodules (cryptoauthlib and FreeRTOS-Kernel), so if you opt to download the beta release manually, you would need to download these submodules manually as well.

That is also why in the git clone command we add  `--recurse-submodules` flag.

Thanks for highlighting this, may be in the next release I will consider adding the submodules as part of the release, or have two releases, one with the submodules and one without.

descartes, 2021-02-28:
To my elderly & simplistic mind, a release archive is complete.

If not, fair enough, then highlighting very clearly that the user has other archives to download. This will save you hours of support queries.

I'm always happy to download submodules if I'm prompted to, but I personally will be doing most of my work against a defined release rather than the moving target of an active repository.

closed 2021-02-28 by descartes

elsalahy, 2021-03-01:
@descartes I agree with you, this can be improved, we can add a documentation, or add two releases.
Since we are in the beta phase, every feedback is extremely welcomed, and will make sure we streamline this as we go forward.