TheThingsIndustries/generic-node-se #133: Simplify application configurations

elsalahy, 2021-03-01:
**Summary:** 

We need to make easy to configure the application layer functionalities.

**Why do we need this?**

To reduce the complexity of building an application with a specific configuration

**What is already there? What do you see now?**

Each app contains `app_conf.h` which enables and disables specific features.

**What is missing? What do you want to see?**

Open for discussion.
Could be:
- A GUI configurator
- A JSON based configurator similar to Mbed OS 
- A simple header file generator
- A simpler `app_conf.h`

**How do you propose to implement this?**

Discuss the options and purpose an example solution.

**Environment:**

BareMetal, OS based

**Acceptance Criteria:**

Should not affect applications targeted functionality

**What can you do yourself and what do you need help with?**

We need to discuss the possible options and I can implement the needed feature.

azerimaker, 2021-03-13:
I've tested this new way of configuring the apps and it's way more intuitive than the previous versions. Good work, @elsalahy .

NicolasMrad, 2022-06-23:
any updates here?

closed 2022-06-27 by azerimaker