TheThingsIndustries/generic-node-se #134: Device Certification

azerimaker, 2021-03-03:
**Summary:** 

Umbrella issue for device certification related issues and discussions. 

**Why do we need this?**

Certification is on the critical path before the mass-market release of our product. 

**What is already there? What do you see now?**

Currently we've started shipping the first [beta] units to our stealth-community members as a sample units w/o certification, but in upcoming months. 

**What is missing? What do you want to see?**

FCC, CE and LoRaWAN certifications. We already have an RF shield in place and we followed many of the design guidelines from the chip manufacturers, so passing the certification shouldn't be too much trouble. 

**How do you propose to implement this?**

1. Firstly, we need to decide whether we want to proceed to the certification with the current layout (v1.1) or wait for the next rev (v1.2). 
2. In any case we can to go through pre-certification to **verify** the current design (v1.1).  
3. Once we have some experience about the performance of the v1.1 and after collecting some feedback from the beta testers we should implement the final layout improvements and go through the certification. Depending on our decision on the list item **1** we can either inherit the certification from the (v1.1) to (v1.2) and do an iterative certification based on similarity or just go directly with (v1.2) as a first-time certification.
4. We have to decide which RF consulting company to work with.  

**Environment:**

Typically an anechoic chamber is used to carry out the certification so if we can find a contract manufacturer with that capability we can save some cost there. 

**Acceptance Criteria:**

Since the current version it's a multi-band device, we have to make sure it complies with all the regional regulations. Plus CE certification will be updated in summer, so we have to account for that too. 

**What can you do yourself and what do you need help with?**

I'll do basic EMI tests with my limited HW setup (scope), especially to verify the DC-DC converter's behavior.

johanstokking, 2021-03-03:
I think it's good to go with LoRaWAN pre-certification with the current revision already. Is the LCTT useful for this? See https://lora-alliance.org/lorawan-certification/

azerimaker, 2021-03-09:
@johanstokking it's a good idea. We'll start with CE (RED) certification route and alongside I will look into the LoRaWAN pre-certification. LCTT will definitely come in handy.

elsalahy, 2021-03-16:
# CE certification Todo:
1. #150 
2. #151 
and doing V1.2 updates if the pre-certification run requires major HW changes, see V1.2 hardware label