TheThingsIndustries/generic-node-se #137: Support for ATECC608A-TNGLORA randomizer

elsalahy, 2021-03-09:
**Summary:** 

We can add support for ATECC608A-TNGLORA random generator.

**Why do we need this?**

To use it instead of the radio random function

**What is already there? What do you see now?**

The stack uses the random radio function

**What is missing? What do you want to see?**

Use the ATECC608A random generator

**How do you propose to implement this?**

Use Microchip cryptoauthlib APIs.
Update `SecureElementRandomNumber()` to use the ATECC608A random generator.

**Environment:**

Baremetal and OS based

**Acceptance Criteria:**

- OTAA joins are operational

**What can you do yourself and what do you need help with?**

All

elsalahy, 2021-05-31:
@marnixcro can you please undertake this?
Should be simple and easy to do, also low priority.

marnixcro, 2021-06-02:
@elsalahy I had a question about this. I currently have some firmware, but don't know if I'm on the right track. What I have know does something like this:
```
    ATCADevice device = atcab_get_device();
    if (atRandom(ca_cmd, &packet) != ATCA_SUCCESS)
    {
        return 0;
    }
    if (calib_execute_command(&packet, device) != ATCA_SUCCESS)
    {
        return 0;
    }
```
I don't know enough about cryptauthlib to know if these commands should be run, so though I should ask before submitting a non-functioning PR. But both commands return ATCA_SUCCESS and the output packet data seems to be random, even when preloading it with zero's (but that does not say that much tbh as it could be random gibberish instead of the randomizer).

If this is the right way, I had another question as the first byte of the return data is always 0x23 (35) but I don't know where this comes from (following bytes are properly random). Couldn't find any mention of this in the datasheet.

elsalahy, 2021-06-02:
@marnixcro please use `atcab_random`, see this [header](https://github.com/MicrochipTech/cryptoauthlib/blob/70d082c0110086ef97d50faa43a9a3c847b4bb12/lib/atca_basic.h#L178)

PR #192 (2021-06-09): SE randomizer support
**Summary:**

This adds the secure element's randomizer for generating random numbers. Additionally adds tags to gitignore. Closes #137, #191.

**Changes:**

- Adds randomizer functionality, which also makes sure that the SE is on as it would otherwise be stuck
- Replaces the instance the randomizer was needed (previously used with radio randomize feature)
- Added 'tags' to gitignore

**Notes for Reviewers:**

Best is to print out the resulting 'rand_in' variables and see these change. Of course, it should join with the OTAA as well.
closed 2021-06-09 by marnixcro