TheThingsIndustries/generic-node-se #145: Improve DC-DC converter decoupling

elsalahy, 2021-03-16:
**Summary:** 

@azerimaker  please update description

azerimaker, 2021-05-05:
In this version, bulk input 100uF capacitors will be replaced with the `GRM31CR60J107ME39L ` which should provide better decoupling.

peterl3233, 2022-09-27:
Hi ... I am working on a design with Murata 1SJ that will use RP605 as well for similar reasons as you.  Would you mind to comment on the necessity to use 3x100uF capacitors at the input of the switch regulator.  I listened to your comments on the YouTube talk and you indicate that its for Hi-Z batteries.  I'm wondering if you still think that much capacitance is necessary - they are expensive and big and i'm just trying to lower cost.

azerimaker, 2022-10-14:
Hi @peterl3233 , 

sorry for the late response. In our case the decision was based on the fact that users might install battery with higher internal resistance (either low quality or highly drained). Therefore, having a bulk capacitance to buffer some of that energy is necessary. If you're using a quality battery in your design having 0805 (20ish uF) decoupling would be enough. 
Please let me know if I should close this issue?

peterl3233, 2022-10-18:
Thanks for the response. And so I figured!   Anyway I have used the lower values and if I see any issues, Ill post to assist others. So .. we're done with this topic.