TheThingsIndustries/generic-node-se #149: Create Enclosure visuals

elsalahy, 2021-03-16:
**Summary:** 

@azerimaker and @wienke  will update the description

wienke, 2021-03-19:
@pierrephz for the Generic Node we need a design for the top and bottom of the device:

* Top - clean visual and there is a button position somewhat at the center.
* Bottom - this should have our company info, CE mark, place for a QR code. And an instruction where people can scan it. Use tti.fyi/qr for now.

azerimaker, 2021-03-25:
@pierrephz I've added dimensions for the front/back panel visuals. Please use this as a reference. 

[GNSEv1.1_label_dimensions.pdf](https://github.com/TheThingsIndustries/generic-node-se/files/6207153/GNSEv1.1_label_dimensions.pdf)

pierrephz, 2021-03-30:
@azerimaker please can you send me the final content/info, CE mark, QR code, etc. when you have it?

azerimaker, 2021-04-12:
@pierrephz hopefully next month we'll have a right to use the CE marking, in the meantime you can already start the design using this resources: 
https://ec.europa.eu/growth/single-market/ce-marking_en
 
Please note that the CE marking must be **no smaller than** **5mm.**

azerimaker, 2021-07-05:
We've picked [this ](https://drive.google.com/file/d/1XM6PYIHjcL4YFFjKD1V_C2ZlUPKB8d2l/view?usp=sharing)design and it will be screen printed with UV resistant color and glue. 
[image]

closed 2021-07-05 by azerimaker