TheThingsIndustries/generic-node-se #157: Option to bypass buck/boost stage

hallard, 2021-03-24:
**Summary:** 
It could be interesting to have the ability to totally bypass the buck/boost stage.

**Why do we need this?**
To be able to power the node with classic Li Socl2 battery and avoid any related quiescent current 

**What is already there? What do you see now?**
I don't see any option to bypass Buck/boost stage

**What is missing? What do you want to see?**
An option to bypass Buck/boost stage

**How do you propose to implement this?**
Build this feature with solder pad/jumper switch or switch

**Acceptance Criteria:**
If implemented with jumper pad (disabled by default), should not have any price incidence

elsalahy, 2021-03-25:
@azerimaker care to provide some input here and is this something we should consider for v1.2 hw version?

azerimaker, 2021-03-25:
@hallard thank you for your suggestion. 
You're right, unfortunately there's no easy way to bypass the buck-boost converter in this version. 
We're planning to remove the voltage supervisor IC and solely rely on the internal UVLO circuitry of the RP605x, therefore in the next rev, we will add a NC-NO jumper pad to be able to disable the IC ([~10nA standby current](https://www.n-redc.co.jp/en/pdf/datasheet/rp605-ea.pdf)) and supply the system directly from the VCC pin, or just add another jumper accross the buck-boost and supply it from the +BATT pin. 

[image]

hallard, 2021-03-25:
Nice shot the CE1 with solder pad of the buck/bust, great idea and 10nA does not hurt me.
Take care powering from VCC may have current draw going to VOUT1 of the buck boost, depending on technology I already had bad surprise with some regulators, so this need to be tested to be sure no current goes to VOUT1, to be sure I would add a solderpad on the output of the regulator (or jumper pin)
On user view (because I can use iron solder without any problem) may be a little 2 position / 2 way switch could be easier to use to select "Use Buck/Bust" or "Straight battery".
I apologize not have been into this project sooner, I can understand it may be late for changes.

azerimaker, 2021-03-25:
We're using a variant of the RP605 with Auto-discharge circuitry (only option available with 2.8V output), so indeed, disabling the IC and feeding power directly from the VCC line might introduce some unwanted leakage paths. I'll verify how much is the leakage with this version and follow up later. 
[image]

But in the next version, we're going to get a variant without this circuitry and looking at the simplified output circuitry we might be able supply the device from VCC without any surprise, since SW1 and SW4 will both be in OFF states. This of course also will need to be verified. 
[image]

hallard, 2021-03-25:
@azerimaker thanks for this awesome and precise answer. That's perfect.