TheThingsIndustries/generic-node-se #158: Is Load Switch 2 for Macronix Flash really necessary?

hallard, 2021-03-24:
Load Switch 2 fir powering SPI Flash (when off) quiescent current can be 10 (typical) to 250nA (max)  and the Macronix flash in power down goes from 7nA (typical) to 350 (max). May be an option to gain some component and remove this one.
The con is to be sure to place the flash in power down when finished job, but it's also an interesting option, never used load switch with flash on my boards.

The Load Switch 1 is mandatory, not all I2C components may have descent power down mode.

elsalahy, 2021-03-25:
@hallard We asked ourselves the same question in the design process and it all comes down to the vision of having GNSE as a boilerplate.

From a FW perspective the external flash can be any external SPI flash (any external flash supporting SFDB will do the trick), so Macronix is a good option when it comes to low power consumption.
The load switch provide assurance that any component that is connected on the SPI bus can be power gated, no matter if it is a component that sleeps well or not.

The cost can be reduced according to customer needs and modifications/optimizations are welcomed.

elsalahy, 2021-03-25:
I think that would be a great topic to discuss in the [Forum](https://www.thethingsnetwork.org/forum/c/nodes/generic-node/88)

hallard, 2021-03-25:
Makes sense also, in all case would provide assurance device is disconnected :-)

By the way, may be you could add this VCC_flash (output of the switch) going to expansion port, this mean that we would be able to shutdown external sensors (or other such as GPS) populated on the expansion board without need to reproduce all power off circuit on our expansion board (and use another GPIO)

azerimaker, 2021-03-25:
Dear @hallard, these are really valuable comments. Thanks a lot. 

> By the way, may be you could add this VCC_flash (output of the switch) going to expansion port, this mean that we would be able to shutdown external sensors (or other such as GPS) populated on the [...]

We also thought about this, in fact on the very first prototype, the I2C expansion was fed by the second load switch (VCC_Flash), but we were afraid that this might cause some confusion if a user wants to add an external sensor and everytime they have to call power-down routine of the flash IC. 

Load switches don't add much to the BOM and in the next rev. we will certainly consider adding one only for external use.

hallard, 2021-03-25:
Awesome, thanks, yes that's correct, always need to think on users side :-)

elsalahy, 2021-04-06:
@hallard is it ok to close this issue?

elsalahy, 2021-06-09:
Closing this issue, please re-open if necessary.

closed 2021-06-09 by elsalahy