TheThingsIndustries/generic-node-se #160: Temperature certification

marnixcro, 2021-03-30:
**Summary:** 

The GNSE might have a use case for the food industry, as a temperature sensor for storage of food in transport, storage and distribution of food. To make it applicable for such a use case, certain reguluations, the [NEN norm](https://drive.google.com/file/d/1AvDF1tmegurT-abZgaKhWCY7nkcNtUON/view?usp=sharing), has to be passed for it to apply.

**Why do we need this?**

For use cases of the GNSE in the food storage industry.

**What is already there? What do you see now?**

GNSE already has temperature sensors, OTA communication (with future FUOTA support), feedback with LEDs and buzzer, etc.. This can already be a start to test certain certifications.

**What is missing? What do you want to see?**

A clear understanding if the GNSE could be used in these use cases, or if it could pass with minor changes to the design.

**How do you propose to implement this?**

Use the NEN-certification and see if the GNSE could pass these, or if it could pass with only minor design changes.

**Environment:**

GNSE

**Acceptance Criteria:**

The GNSE could pass certification (maybe with minor design changes) or it could certainly not pass the certification.

**What can you do yourself and what do you need help with?**

All

marnixcro, 2021-04-01:
# Checklist for temperature sensor application

This are a brief summary of the things that we need to do before the GNSE is ready to be a capable temperature sensor for industry use. Note that all tests are listed in chapter 6. I'll continue updating this list, and will include a separate checklist for all specific tests as these are a bit more specific (and way longer).

## Cloud (TTN and possible extensions)
- [x] All data is traceable (audit trail), timestamps and origin device should be attached to all data.
- [ ] The traceable data is kept for at least a year (if we want to do frozen food).
- [x] Data easily accessible by (authorized) users.
- [ ] Give indication that the battery is low (could also be locally with buzzer, but this seems easier)

## Hardware (enclosure and PCB)
- [ ] IP rating should be tested. The higher the IP, the more applications we could use the GNSE for (chapter 5.7).
- [ ] Temperature range should be tested (see chapter 5.10). SHTC3 should be within the best class, but you never know how the enclosure will influence this.
- [ ] The response time for the sensors in casing should be less than 20 minutes.
- [ ] Vibration testing (chapter 5.10.3), but only if we think that transportation is a good use case (might not be with LoRa, but who knows). 

## Software
- [ ] Adjustable interval/limit settings (optionally, but preferably, with a downlink)
- [ ] Transmit the battery data as well
- [ ] Read Annex D for the periods the sensors have to gather data at least (these all well within reachable limits, but still good to note when designing the software).
- [ ] Software has to apply to a certain compliance based on the system (chapter 5.12)

## Misc
- [ ] Look into a few electrical power certifications: EN 61010-1 EN 61010-2, and EN 61010-3  These are only vaguely mentioned a few times (chapter 5.8 and 5.9). I don't think it is really for us (as we use low DC voltages) but maybe there is another important thing we did not note.
- [ ] Look if timing errors (chapter 5.10.2) are an issue for us (can't image it to be an issue, but still).
- [ ] If we comply, the marketing should note this as well (chapter 8).

marnixcro, 2021-06-29:
@wienke how should we move forward with this? Update the missing parts and continue, wait it out, etc.?

wienke, 2021-07-12:
Let's focus on the basic app right now and put this to the backlog.

elsalahy, 2021-08-20:
@marnixcro could you please provide a small report on the progress done on this issue from your side.

The report should be:
1. In PDF format, and uploaded [here](https://drive.google.com/drive/folders/17DJXbyA3pLFlHnEMJizYBwRDDytM6GVK?usp=sharing).
2. Should be concise detailing the requirements, your analysis and achieved results.

elsalahy, 2021-10-12:
This issue was moved to our discuss.