TheThingsIndustries/generic-node-se #163: Antenna Performance w/ and w/o enclosure.

ehogeweg, 2021-04-01:
Hi,

I noticed that the enclosure screws are close to the antenna clearance area. I expect this has some impact on the tuning frequency. Is there any data, that you can share, on how the antenna performs on a bare board vs. enclosed? Does the matching network require different component?

Cheers,

Erwin

azerimaker, 2021-04-12:
Thank you for you interest, @ehogeweg. 
I understand your concern. In fact, we've discussed this even before we had the PCBs at hand in issues #89, #128. 
The enclosure and the close proximity of the screws has certainly some impact on the ANT performance, however we don't have any comparison data or quantitative info on that at this moment. 
We're working on it and once we have it, I'll share our findings here.

azerimaker, 2021-05-05:
@ehogeweg , @lnlp I did a basic efficiency test using my NanoVNA. Please see the results here ➡️  https://github.com/TheThingsIndustries/generic-node-se/issues/89#issuecomment-833003749
You're right, the enclosure and the batteries do affect the antenna performance and removing it results in shifting the tuning frequency upwards. In the coming weeks the device will go through proper matching network retuning. Hopefully, this will provide good performance with the enclosure on.

elsalahy, 2021-07-05:
@azerimaker did Ignion provide any feedback on the effect of the enclosure on the antenna performance?

azerimaker, 2021-07-05:
> @azerimaker did Ignion provide any feedback on the effect of the enclosure on the antenna performance?

unfortunately their "Matching Network Optimization" service doesn't include that, but I will ask them to perform those tests, hopefully they agree to that.

azerimaker, 2021-07-07:
@lnlp Ignion has conducted the above mentioned tests and it's clear that frequency shift is major. One should use the device as a fully assembled unit if they want to achieve good coverage.  I hope this answers your question. If there are no further discussion, please consider closing this issue. 

[image]

[image]

azerimaker, 2021-07-07:
> For positioning of the device it will be interesting to know the radiation pattern and how the device can be best positioned for optimal RF coverage/range. _(The battery holder and batteries will block, reflect and deflect RF signals.)_

Unlike anticipated the batteries and the contacts didn't have major impact on the antenna performance and its radiation pattern. Please see the results below for your reference. 
 
[image]
[image]
[image]

elsalahy, 2021-07-07:
@ehogeweg  Does the above answer provide sufficient information for GNSE use with an enclosure.
If yes, could you please the issue

ehogeweg, 2021-07-09:
Sorry, I completely missed your comments. Yes, this answers the question. Thanks for providing the graphs.

Cheers,

Erwin

closed 2021-07-09 by ehogeweg