TheThingsIndustries/generic-node-se #175: Payload buffering in the device

wienke, 2021-04-19:
**Summary:** 

We would need a way for data that is measured that can be buffered in the device if it did not arrive.

**Why do we need this?**

...

**What is already there? What do you see now?**

...

**What is missing? What do you want to see?**

...

**How do you propose to implement this?**

...

**Environment:**

...

**Acceptance Criteria:**

...

**What can you do yourself and what do you need help with?**

...

HeadBoffin, 2021-05-16:
This topic has many discussions on the forum with many different approaches, many designs become problematic for FUP for downlinks.

elsalahy, 2021-07-05:
@marnixcro could you please take the lead in designing and implementing this library.

elsalahy, 2021-08-20:
@marnixcro have you done any development for this issue, if yes, please share the branch.