TheThingsIndustries/generic-node-se #178: GNSE sensors sampling app (vanilla version)

elsalahy, 2021-04-26:
**Summary:** 

We need to build an application showing sampling board sensors and sending the data over LoRaWAN.

**Why do we need this?**

To show board usage and have foundation firmware for production and certification.

**What is already there? What do you see now?**

Basic apps.

**What is missing? What do you want to see?**

An advanced low power application that samples the data and sends information over LoRaWAN.

**How do you propose to implement this?**

Using basic apps, doing power analysis and using HAL APIs

**Environment:**

Baremetal

**Acceptance Criteria:**

- Low power 
- Compiles with no warnings
- Extremely configurable
- Used GNSE HAL

**What can you do yourself and what do you need help with?**

All

descartes, 2021-05-14:
Having just answered a question on the forum about the existence of the payload formatter for what we are assuming to be the pre-loaded firmware, I have to express some considerable surprise that there isn't a simple example reading one of the sensors and sending it - this to me would be the most blindingly obvious hello-world app for a LoRaWAN device and it's whole raison d'être!

elsalahy, 2021-05-18:
@descartes Yes agreed.
We should prioritize this issue.

descartes, 2021-05-20:
Looking at your original text again, I'd recommend having three versions - one that just reads a sensor & sends the data, then an incremental version that adds in low power / sleep, then what ever meets the criteria for 'advanced'.

I'd shy away from "extremely configurable" as that usually infers either lots of #defines or lots of OOP or both and then people have trouble tracing the code and get confused.

elsalahy, 2021-06-29:
This issue is partially addressed in #194 
The completion of this issue depends on #171

elsalahy, 2021-08-25:
This is closed by #205

closed 2021-08-25 by elsalahy