TheThingsIndustries/generic-node-se #181: Mbed OS build environment integration

marnixcro, 2021-04-29:
**Summary:** 

Mbed OS provides multiple [development tools](https://os.mbed.com/docs/mbed-os/v6.10/introduction/tools.html) to build the project. MBed can also be build with [a CMake environment](https://os.mbed.com/cookbook/mbed-cmake), similar to our CLI setup.

**Why do we need this?**

Mbed OS has to be compiled in some form and will require some work to be properly integrated into our design.

**What is missing? What do you want to see?**

No support for Mbed.

**How do you propose to implement this?**

Add Mbed through the mbed-cmake example or the Mbed development tools.

**Environment:**

Mbed

**What can you do yourself and what do you need help with?**

All

marnixcro, 2021-05-31:
Mbed OS can be build in (private) repo using Mbed CLI 2 and altering the application header file to switch different applications.

azerimaker, 2022-06-27:
Mbed is is not a prirority ATM, therefore I think it's best we close this topic now and reopen it when needed.