TheThingsIndustries/generic-node-se #182: Mbed LoRaWAN example

marnixcro, 2021-04-29:
**Summary:** 

Add an application utilising LoRaWAN and Mbed OS.

**What is already there? What do you see now?**

LoRaWAN examples with FreeROTS, AzureRTOS and bare metal.

**What is missing? What do you want to see?**

Mbed OS integration with LoRaWAN.

**How do you propose to implement this?**

A [LoRaWAN example](https://github.com/ARMmbed/mbed-os-example-lorawan) and [STM32WL support](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32WL) already exist. These, with additional pin mapping and GNSE firmware, should be integrated into an application for LoRaWAN.

**Environment:**

Mbed

**Acceptance Criteria:**

Application with Mbed works with LoRaWAN.

**What can you do yourself and what do you need help with?**

All

marnixcro, 2021-05-31:
Issue was fixed in the (private) GNSE OS mbed firmware application. [Standard LoRaWAN example](https://github.com/ARMmbed/mbed-os-example-lorawan) can work as well.

elsalahy, 2021-07-12:
Still in review, moving this to Q3

NicolasMrad, 2022-06-23:
any updates here?

azerimaker, 2022-06-27:
Not much, please see the other comment.

closed 2022-06-27 by azerimaker