TheThingsIndustries/generic-node-se #183: Mbed OS-GNSE integration

marnixcro, 2021-04-29:
**Summary:** 

Umbrella issue for integrating Mbed OS with GNSE

- [ ] Integrate Mbed build environment #181 
- [ ] Get LoRaWAN functioning #182

elsalahy, 2021-05-27:
@marnixcro what is the status and how can I assist?

marnixcro, 2021-05-27:
> @marnixcro what is the status and how can I assist?

I added the changes from [this commit](https://github.com/TheThingsIndustries/mbed-os/commit/cc48c6f3f5c57d957a72d12894ef6e944d49ed3c) and got it to compile the lorawan example, but I still need to follow [your guide](https://github.com/TheThingsNetwork/lorawan-example-atecc608a-tnglora/tree/master/mbed-os) to get it working (I had some build conflicts the first time around, but saw what I did wrong). I think it could be done before the weekend, otherwise I'll let you know if I hit a wall somewhere. I'll also update the issues on git with the secure element.

marnixcro, 2021-06-29:
@elsalahy can you please review the [application](https://github.com/marnixcro/mbed-os-generic-node-se) and the changes in the branch.

elsalahy, 2021-06-29:
@marnixcro I will review the repository and give my feedback within this week.
anything in specific should I focus on?

marnixcro, 2021-06-29:
> @marnixcro I will review the repository and give my feedback within this week.
> anything in specific should I focus on?

Maybe kinda the UX (for things such as switching the applications, does that make sense from a user perspective). Of course regular things like code/git-hist cleanliness, functionality, etc. are still important.