TheThingsIndustries/generic-node-se #185: Will rel 1.1 dev board work in US?

ehogeweg, 2021-05-11:
Hi,

Is there anything else, besides setting ACTIVE_REGION to LORAMAC_REGION_US915 I need to do to make the v1.1 GNSE work here in the US?

How is the antenna matching adjusted for US freq. range? It may be in the docs but I couldn't find it.

Regards,

Erwin

elsalahy, 2021-05-18:
@ehogeweg Thanks for your question.

Indeed Adjusting `ACTIVE_REGION` to `LORAMAC_REGION_US915` will allow the device to function in the US.

@azerimaker can provide more information on the antenna matching and the node support for 868 and 915 frequencies.

azerimaker, 2021-05-20:
@ehogeweg thank you for your question. 
Default matching network is actually better suited for ~915MHz range, so it will certainly work in the US. I did a basic antenna return loss analysis and you can read more about our findings here ➡️  https://github.com/TheThingsIndustries/generic-node-se/issues/89#issuecomment-833003749

closed 2021-05-20 by azerimaker