TheThingsIndustries/generic-node-se #186: Out-the-box firmware source code does not appear to be available

descartes, 2021-05-14:
**Summary:** 
We don't know what the default firmware is doing or sending

**Steps to Reproduce:** 

1. Obtain node
2. Register it
3. Turn it on
4. Look at blinkenlights
5. Admire uplink in console
6. Not have clue what uplink means

**What do you see now?**
A payload with no meaning

**What do you want to see instead?**
Preferably the source code, from which I can create a payload formatter

Or documentation on the payload so I can create a payload formatter

Or a payload formatter

**How do you propose to implement this?**
Using my Ninja Payload Formatting skillz - or more likely, my Payload Formatter spreadsheet that I use to create Payload Formatters semi-automatically

**What can you do yourself and what do you need help with?**
Depends on what resources you provide!

I can also add this to the device repository as well.

elsalahy, 2021-05-18:
@descartes 
The 100 batch was shipped with basic app as a default FW, and it prints serial information using UART at 115200 8N1.

I will prioritize #178 sensors vanilla app as this will be the application that will be used in the next batch.
And I can provide a simple payload formatter or provide some support to you in case you want to take this on.

For now, this is pending on #178

descartes, 2021-05-20:
> And I can provide a simple payload formatter or provide some support to you in case you want to take this on.

I'm happy to create a simple starter app and payload formatter if that's what you are suggesting. If it's just the payload formatter, I have an app for that.

descartes, 2021-05-20:
PS, I need a simple something running for the mini-conference at the end of next week for my talk on WebHooks, so I will be creating a sensor read & send example anyway.

azerimaker, 2021-05-20:
I've already replied on the [forum](https://www.thethingsnetwork.org/forum/t/box-opening-no-video/45548/26?u=cndrxn), just adding here for redundancy. 
@descartes if you want something quick and dirty you can check out this [branch](https://github.com/TheThingsIndustries/generic-node-se/blob/feature/basic_lorawan_TH/Software/app/basic_lorawan_th/lora_app.c), which sends temperature and humidity values. 

Simple payload decoder I used: 
```
function decodeUplink(input) {
  var data = {};
  data.temperature = ((input.bytes[0] << 8) + input.bytes[1]-400)/10;
  data.humidity = ((input.bytes[2] << 8) + input.bytes[3])/10;
  return {
    data: data,
  };
}
```

elsalahy, 2021-05-20:
@descartes check @azerimaker [comment](https://www.thethingsnetwork.org/forum/t/box-opening-no-video/45548/26?u=elsalahy), maybe this is a good simple example for you.

descartes, 2021-05-20:
All good, rest assured as a moderator I read every single post, all of them - mostly skimming them for guidelines, but still, redundancy for the win!

descartes, 2021-05-20:
The payload formatter you have is a v2 style which still works, but I'll update it / create something to run on v3. I cut my teeth on LiveScript back in Sept 1995 so my biggest problem now a days is typing the right programming language in to the right IDE.

elsalahy, 2021-06-29:
This issue is partially addressed in #194
The new patch of devices will have the `sensors_lorawan` application  communicated in #178.

@descartes is your issue addressed in the new application `sensors_lorawan`?
If yes, is it possible to close this issue?

descartes, 2021-07-01:
Yes, gives something sensible as a starter and pretty much the same as the code I used for my TTS conference demo on WebHooks.

Sleep will need to be added very soon - the most basic of nodes usually reads & sends sensor data & then sleeps for a chunk of time.

closed 2021-07-01 by descartes