TheThingsIndustries/generic-node-se #188: Road to circular IoT device

wienke, 2021-05-21:
**Summary:** 
Inspired by this great new initiative (https://cep2030.org) this thread is focussed on discussion around how to include circularity by design.

**Why do we need this?**
We need to work towards a more circular product. The big IoT promise can not come at the cost of a huge amount of environmental waste. We are smart and resourceful enough to make products that are both more sustainable and durable.

**What is already there? What do you see now?**
Looking at the current design we already implemented some features that should make the product more sustainable.

- We focussed on a power management design that is able to optimize the use of alkaline batteries. Assuming that most places in the world already have a circular economy for this type of batteries. Also it is the most common battery. The chance of somebody getting rid of the product once batteries need to be replace because it uses some kind of exotic industrial battery is lowered. Also through a hardware switch the device can be switched to using 3v batteries (violates the certification)
- The design is modular. You are able to upgrade or repurpose the device later on by extending it with sensors, changing the casing, upgrading the firmware and changing the radio technology used through firmware.
- We added a piezo speaker. LPWAN devices are typically installed out of sight. To re-purpose the device you first have to find it. By sending it a commend to play a tune you will be able to find it.

**What is missing? What do you want to see?**

So there are many more design patters we should be able to implement.
- Implementing accountability. The device is equipped with a Secure Enclave. So the device will have a uniquely traceable identity. This also allows for the transfer of ownership. And also allows for administrating accountability to that owner for re-cycling and re-purposing. Adding a deposit model to this might bring the right incentives as well.
- Creating a second hand market that is generating demand for re-purposed IoT sensors.
- Creating a license model that discounts usage in proper circular ways.

elsalahy, 2021-07-01:
@azerimaker care to comment on this proposal.

@wienke what should be the outcome of this issue, and should we move this discussion to our Discuss forum?

wienke, 2021-07-07:
This is indeed a place holder for now. We can move it to the backlog.

azerimaker, 2021-07-07:
@wienke (https://cep2030.org) is indeed a great initiative. IMO our GN doesn't really fit into their initial product categories (temperature exchange equipment, screens and monitors, lamps, large equipment, and small IT) we can still consider planning ahead for reducing e-waste. Some big players like Apple have full control over the production and they can control the whole supply chain, push component manufacturers to produce components only from recycled materials. 

Using recycled materials in the design, making it more repairable are good design choices, but heavily emphasizing it for marketing gain is not so much.

TheThingsBot, 2021-09-24:
This issue has been mentioned on **Discuss The Things Industries**. There might be relevant details there:

https://discuss.thethingsindustries.com/t/road-to-circular-iot-device/658/1

elsalahy, 2021-09-24:
Any discussion on this topic should take place [here](https://discuss.thethingsindustries.com/t/road-to-circular-iot-device/658)

closed 2021-09-24 by elsalahy