TheThingsIndustries/generic-node-se #196: Broken multicast support (McRootKey is always generated based on LoRaWAN version 1.1.0 scheme)

elsalahy, 2021-06-16:
**Summary:** 

Our current LoRaWAN stack generates the MC keys based on 1.1 even when configured for 1.0.3

This is a mirror of an already fixed issue on LMN version 4.4.5

https://github.com/Lora-net/LoRaMac-node/issues/936

elsalahy, 2021-06-16:
@johanstokking we use an ST LoRaWAN stack that is based on LMN v4.4.4

A lot of critical fixes related to FUOTA and LoRaWAN have been done since then as you can see from the [change log](https://github.com/Lora-net/LoRaMac-node/blob/master/CHANGELOG.md)

Should I just keep patching our current stack?
The issue mentioned above wasted a lot of my effort o debug and identify the root cause.

johanstokking, 2021-06-19:
> Should I just keep patching our current stack?

What is the alternative? Switch to LMN and add ST support ourselves?

elsalahy, 2021-06-21:
> What is the alternative? Switch to LMN and add ST support ourselves?

Yes
This approach also has it's disadvantages, but it pays out in the long term.

As a short term solution I can keep patching the current ST version until they fix it them selves.

johanstokking, 2021-06-22:
What is the balance in terms of effort between to these two approaches?

1. Take vanilla LMN and add ST support
2. Take ST LMN and patch upstream changes

johanstokking, 2021-06-24:
OK, then I'm in favor of going with taking the vanilla LMN and add ST support.

Until, of course, ST publishes a stable version that covers our use cases. I don't see this happening in the short term, especially not with LoRaWAN 1.0.4 and 1.1.1 coming up.

elsalahy, 2021-06-25:
Thanks @johanstokking.
I will plan this accordingly.

PR #197 (2021-06-29): Feature/basic fuota
**Summary:**

Adds basic fuota app and server
The application attempts to handle an interoperability fuota test.
closes #69 
closes #82 
closes #196 

**Changes:**

- A number of changes to the stack and the LmHandler
- Adds a basic_fuota app

**Notes for Reviewers:**

Please observe the video I shared on slack.
@johanstokking do you think I should add the JS test server within this application, or just point to [this repo](https://github.com/elsalahy/test-fuota-server) in the documentation.

**Release Notes: _(optional)_**

...
closed 2021-06-29 by elsalahy