TheThingsIndustries/generic-node-se #203: FreeRTOS RX retrieval windows to long

marnixcro, 2021-07-05:
**Summary:** 

FreeRTOS has some drift, causing downlinks to fail. This relates to PR #202 and issue #187.

**Steps to Reproduce:** 

1. Default value should show this in current consumption:
[image]
2. Change this:
https://github.com/TheThingsIndustries/generic-node-se/blob/40636650a7cb9c0030e3905d1359de02d75471b3/Software/app/freertos_lorawan/conf/lorawan_conf.h#L117
And this will produce this image:
[image]
With this timing:

|basic_lorawan |freertos_lorawan | (new) freertos
|---|---|---|
|5s14ms | 5s129ms |4s980ms|
|218ms|271ms |531ms|
|781ms | 754ms | 492ms|
|218ms | 271ms |531ms| 

**What do you see now?**

Delayed RX windows.

**What do you want to see instead?**

RX windows closer to their appropriate values.

**How do you propose to implement this?**

Find the cause of the drift delaying the downlink.

**Environment:**

FreeRTOS

**What can you do yourself and what do you need help with?**

All

elsalahy, 2021-08-20:
Can you please share some informations on your attempts to solve this?

elsalahy, 2021-08-20:
@marnixcro Can you please share some informations on your attempts to solve this?

marnixcro, 2021-08-20:
> @marnixcro Can you please share some informations on your attempts to solve this?

I tried finding the cause with the clocks drifting, but the datasheet mentioned max drifts of only 1-2% on these clocks. I have not been able to find where in the firmware the downlink event gets triggered.