TheThingsIndustries/generic-node-se #204: Jitter and backoff implementation for basic_lorawan

marnixcro, 2021-07-07:
**Summary:** 

Most of the LoRaWAN applications retry join attempts after a set amount of time (default is around 10s). This can lead to problems when large deployments are reset, as it can hog the network, with the potential of none of the devices being able to join. This can be improved by adding jitter to randomize join attempts. Additionally, backoff can be implemented as well. FreeRTOS [already implements jitter](https://github.com/TheThingsIndustries/generic-node-se/blob/40636650a7cb9c0030e3905d1359de02d75471b3/Software/lib/FreeRTOS-LoRaWAN/freertos_lorawan.c#L650) but not backoff. This is related to #106.

**Why do we need this?**

This [is considered](https://www.thethingsnetwork.org/forum/t/does-a-node-ever-need-to-rejoin-after-otaa/2442/12?u=htdvisser) a best practice for LoRa as it helps with use cases with multiple devices.

**What is already there? What do you see now?**

~10s rejoin attempts in regular LoRaWAN applications, with jitter and backoff implementation in freertos_lorawan (even configurable in app_conf.h).

**What is missing? What do you want to see?**

Jitter and backoff implemented to randomize (and constantly increase) rejoin times, optionally configurable in app_conf.h with a max, which was done for freertos_lorawan. It only needs to target rejoins and have a reasonable max value to not interfere with development and regular use cases.

**How do you propose to implement this?**

Use a randomizer function and wait this amount of milliseconds on rejoins (jitter), which increases over time (backoff)

**Environment:**

Baremetal

**Acceptance Criteria:**

Network rejoins don't happen based on a set value, but use jitter and backoff.

**What can you do yourself and what do you need help with?**

All

mluis1, 2021-07-07:
Just for info the MAC layer already implements the Join back-off as specified by the LoRaWAN specifications chapter "7 Retransmissions back-off" (The seen behavior is like the duty-cycle enforcement)
This feature is always enforced.

The only thing that is required at application level is to insure that the end-device transmissions are randomized.

marnixcro, 2021-07-07:
> Just for info the MAC layer already implements the Join back-off as specified by the LoRaWAN specifications chapter "7 Retransmissions back-off" (The seen behavior is like the duty-cycle [...]

You're right, it's not a problem with the mac layer's implementation but rather on the device level application since basic_lorawan will join (after the initial failed attempt) when trying to sent, and the timing is set by APP_TX_DUTYCYCLE (which is set at the mentioned 10s).

elsalahy, 2021-07-08:
@marnixcro can you please adjust the the description of the issue following Miguel's input.
Mainly concerning the need for backoff.

NicolasMrad, 2022-06-23:
is this still relevant?

azerimaker, 2022-06-27:
It is. We need to look into this.