TheThingsIndustries/generic-node-se #206: GNSE/STM32WL supported platforms

elsalahy, 2021-07-12:
| Feature/Platform | STM32WL             | GNSE                           | LoRaWAN Class A Example   |
|------------------|---------------------|--------------------------------|---------------------------|
| FreeRTOS         | :white_check_mark:  | :white_check_mark:             | :white_check_mark:        |
| AzureRTOS        | :white_check_mark: | :white_check_mark:             | :arrows_counterclockwise: |
| Mbed             | :white_check_mark:  | :white_check_mark: (in review) | :white_check_mark:        |
| Zephyr           | :white_check_mark:  | :arrows_counterclockwise:      | :white_check_mark:        |
| RIOT             | :white_check_mark:  | :arrows_counterclockwise:      | :white_check_mark:        |

TheThingsBot, 2021-07-12:
This issue has been mentioned on **Discuss The Things Industries**. There might be relevant details there:

https://discuss.thethingsindustries.com/t/gnse-supported-platforms-official-and-community/583/1

elsalahy, 2021-07-12:
Of course the bare-metal support exists with 4 apps now:
- basic
- basic_lorawan
- freefall_lorawan
- sensros_lorawan

azerimaker, 2021-07-12:
I'd include Arduino to this list as well and make try get community help for its improvement. There's already some work on this: 
https://github.com/stm32duino/Arduino_Core_STM32/pull/1429
https://github.com/stm32duino/Arduino_Core_STM32/pull/1428

hallard, 2021-07-22:
going further on supported platform, @elsalahy I've done some board based on STM32WLE5JCIx (LoRa-E5) and STM32WLE5CCUx (RAK3172), you can take a look on this [post](https://www.thethingsnetwork.org/forum/t/custom-boards-based-on-lora-e5-and-rak3172/49947) for the boards.

How difficult would be to add them on target ? We already added them for mbed-os so I guess it could be easy here also with some starting point I don't have.

- First add the CPU target (single core) for STM32WLE5 and Hardware definition file if any
- Change Radio Driver (the part that control RF switch CTRL1, CTRL2 and CTRL3 according to wiring of these modules)
- change NO_TXCO for RAK3172 since it don't have one.

Any idea are welcome, if first step is done I can try doing 2nd and 3rd.

elsalahy, 2021-07-27:
@hallard the board looks great and I like that you already added official support for Mbed OS.

I can assist with points 2 and 3, can you please share your fork where you did point 1.

The current repo is designed to be compatible with the single core variant of STM32WLE5, so adding support for your board should be easy and straight forward.

hallard, 2021-07-27:
> @hallard the board looks great and I like that you already added official support for Mbed OS.
> 
> I can assist with points 2 and 3, can you please share your fork where you did point 1.
> 

Code is here 
https://github.com/ARMmbed/stm32customtargets/blob/master/TARGET_STM32WL/TARGET_LORA_E5/LORA_E5_radio_driver.cpp

elsalahy, 2021-07-27:
@hallard This looks great, thanks for the tips and helpful links.
@marnixcro could we do something similar for GNSE (related to #183).

sabas1080, 2021-08-13:
Basic Support ready for STM32duino https://github.com/stm32duino/Arduino_Core_STM32/pull/1459

lnlp, 2021-08-14:
> Basic Support ready for STM32duino

Please don't use the name STM32duino.
It's confusing (which I motivated on several places on the internet).

The correct name for this Arduino core is: `Arduino Core STM32`.

elsalahy, 2021-08-19:
Awesome @sabas1080, do we need to wait for v2.1.0 to try it?
STM32WL is not indexed yet in [here](https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/package_stmicroelectronics_index.json) 

[image]

sabas1080, 2021-08-19:
@elsalahy yes or add manually

fpistm, 2022-12-21:
Hi @sabas1080  and all,

We will provide soon the LoRa support of the STM32WLxx:
- LoRaWan thanks https://github.com/stm32duino/STM32LoRaWAN (Still one PR to merge soon and we will released officially the library)
- LoRaRadio thanks RadioLib and this PR https://github.com/jgromes/RadioLib/pull/649

To use them you will need STM32 core version 2.4.0 which will be released in the coming days.
Any feedback will be appreciate.

sabas1080, 2022-12-21:
Thanks @fpistm, I am working with @lizethgm for make it work RAK3172 and Generic Node TTN  (https://github.com/stm32duino/STM32LoRaWAN/issues/5) and 915Mhz working ;)

fpistm, 2022-12-22:
Thanks @sabas1080 

FYI, STM32 core v2.4.0 is released.