TheThingsIndustries/generic-node-se #208: Configurable sensors transmission interval via downlink

elsalahy, 2021-07-12:
**Summary:** 

It should be possible to adjust `sensors_lorawan` transmission interval via a downlink.

**Why do we need this?**

To be able to change the device behavior using a downlink. 

**What is already there? What do you see now?**

A programmed transmission interval (can be configured before compilation/flashing).

**What is missing? What do you want to see?**

Adjusting the transmission interval using a downlink.

**How do you propose to implement this?**

- Define a downlink port that should be used for configuration.
- Send a downlink on that port that requests a change of transmission interval.
- The downlink of transmission interval value should be passed to the system scheduler.
- The scheduler should check if the value of the transmission interval is acceptable, if yes, then apply it, if no revert to previous configuration. 

Please note:

- Default configurations should be proposed here in this issue.
- Configurations should be easy to change such as Downlink Fport, default interval,...,etc.

**Environment:**

Baremetal.

**Acceptance Criteria:**

- The app performs as expected after adjusting the interval.
- The device should sleep in-between transmissions. 

**What can you do yourself and what do you need help with?**

Can be done by one person.

elsalahy, 2021-07-12:
@marnixcro please ask any questions you might have here before jumping into implementation.
This issue is a breakdown of the out of the box app wishlist [here](https://discuss.thethingsindustries.com/t/generic-node-sensor-edition-apps/475/5)

elsalahy, 2021-07-12:
@marnixcro please also add iInteractive LED behavior on sensors transmissions (blinking).
But please make sure you sleep properly (set used LED pins to analog and add them in stop mode hooks)

SloMusti, 2021-07-13:
The practice I found useful is for devices is to allow for a system interval to be in range of 1-1440 minutes, which means device will try to send at least one a day and one can not configure it outside that scope. A desired implementation of this setting would also be to be reboot persistent, particularly if a system default is rather fast, as otherwise an unnoticed reboot can result in poor battery life.

marnixcro, 2021-07-22:
@elsalahy some minor questions:
1. You probably also want a led sequence for the downlinks right?
2. I think the idea of it being boot persistent from @SloMusti makes a lot of sense. Would it be better to do it on the external flash or try to do it in the internal one you think? Internal flash would be nicer in my opinion as the user could very easily overwrite it by erasing it before programming, making development somewhat easier. Although the external flash could have other benefits you might prefer.
3. The LED should probably also indicate the downlink right (with different blinks if it passed or failed)?

elsalahy, 2021-07-27:
@marnixcro 

1. Feel free to propose something.
2. Please open another issue for this (reboot persistence) and keep this issue focused on configurable transmission interval via downlinks.
3. Feel free to propose something, I guess successful downlink config should be Green, and failure should be Red!