TheThingsIndustries/generic-node-se #214: Reboot persistence

marnixcro, 2021-07-27:
**Summary:** 

All applications will start anew, meaning it does not matter what the reason for a reset it, it will just reboot. This can cause issues when devices are stationed that have configurations done at run-time (with downlinks for instance) as any reboot can return to their old configs without it being expected by the user. Related to #208.

**Why do we need this?**

It helps with some applications where some configurations should be kept after reboots.

**What is already there? What do you see now?**

Any configs made by the user are not saved at all.

**What is missing? What do you want to see?**

Some configs that can be saved after a reboot.

**How do you propose to implement this?**

All RAM is cleared, so these configs should be stored on the flash (on the MCU or the external flash).

**Environment:**

Bare metal

**Acceptance Criteria:**

Some form of reboot persistence exist for the devices. 

**What can you do yourself and what do you need help with?**

All

elsalahy, 2021-08-20:
@marnixcro have you done any development for this issue, if yes, please share the branch.

marnixcro, 2021-08-20:
> @marnixcro have you done any development for this issue, if yes, please share the branch.

I pushed it to feature/feature-reboot (although there was not a lot of progression on this).