TheThingsIndustries/generic-node-se #215: Basic application for certification

marnixcro, 2021-07-27:
**Summary:** 

Umbrella issue and discussion place for creating an application for testing the hardware of the GNSE for production/certification, similar to the basic app.

elsalahy, 2021-08-04:
The basic app is designed to verify the functionality of all on-board peripherals.
This app can be used in the production line to verify the PCB.

I think we need to add the following:
1. Report failed component (Buzzer, SE,..., etc) with diagnostics code
2. Improve readability of the results (simplify wording, reduce log size)

We can add more improvements/features depending on manufacturing facility feedback.

elsalahy, 2021-09-24:
This was done and sent to the manufacturing facility by combining the basic app with basic_lorawan app.
If further improvements are needed we can re-open this issue.

closed 2021-09-24 by elsalahy