TheThingsIndustries/generic-node-se #218: Design a custom sleeve for the Generic Node (SE) packaging

azerimaker, 2021-08-09:
**Summary:** 

As the GN production is proceeding we need to come up with a sleeve design very similar to one of our earlier product: the Things Node. 

**Why do we need this?**

It needs to be GN branded and highlight its basic features and cloud connectivity. 

**What is already there? What do you see now?**

Previous design of our node could be a good reference point. 
[image]

...

**What is missing? What do you want to see?**

Exact dimensions and what into to print will be updated once we decide the the exact packaging style. 
For now, the following information have to be included on it (cont. updated):

- GN logo
- "_Generic Node Sensor Edition_" name
- TTI logo
- Getting started url: https://www.genericnode.com/docs/getting-started/ (maybe shorter url)
- CE, ESD, Zero Waste symbols. 
- Company and Manufacturing info. (will be defined later)

**Acceptance Criteria:**

The sleeve design must contain all the relevant information and must comply with EU regulations. 

**What can you do yourself and what do you need help with?**

I'll provide all the necessary dimensions and details to aid the design process.

pierrephz, 2021-08-18:
Hey, seeing this issue I think it must be the other way around:

> Exact dimensions and what into to print will be updated once we decide the the exact packaging style.
> For now, the following information have to be included on it (cont. updated):

I have the branding, we have the assets, there is no problem, but I need to know the dimension to create the structure / the base of the work

cc @rish-c @azerimaker

rish-c, 2021-08-18:
Possible dimension - 
20 X 15 X 6 (cm)

Played around a bit and found this website for reference - http://ronenness.github.io/3D-Box-Shot-Maker/

There are 2 ways to have the packaging - 
* Sleeve design^ - so the top cover slides off and you get the tray where the node along with other accessories are there
* Open design^^ - open the box and the parts are stuffed inside the styrofoam packaging (I prefer this one actually)

Possible design options - 
* Front side - GN logo with text "One One LoRaWAN® node, Endless use cases."
* Back side - TTI logo with our PO Box address 
* Side  1 - have these icons 
[image]
* Side 2 - Made by The Things Industries

^ - 
[image]

^^ - 
[image]

What are your thoughts? @azerimaker

azerimaker, 2021-08-20:
@rish-c  20x15x6 cm sounds plausible. 
Since the inner packaging part will be handled by the CM we have to ask them if that size is sourceable, but it will be pretty close to this size. @pierrephz  please work with this dimension to start off. 

If they can get the branding info printed on the box directly that would simplify the design, but last time they mentioned that the sleeve approach would be simpler for them.

NicolasMrad, 2022-03-29:
Can this be closed?

closed 2022-04-19 by NicolasMrad

azerimaker, 2022-04-19:
@NicolasMrad thank you for closing this.