TheThingsIndustries/generic-node-se #248: High current consumption in "stop mode"

ThomasBaumgartner, 2023-04-12:
**Summary:** 

I configured the application with lowpower options. I connected power analyser to the device (Generci Noce V1.1)and measured current consumption during stop mode. The power analyser was indicating a minimum current consumption of 33uA, which is rather high.  

**Steps to Reproduce:** 

1. In app_conf.h set: 
GNSE_ADVANCED_TRACER_ENABLE 0
DEBUGGER_ON       0

2. build and flash sensors_lorawan application
3. Measure the current consumption

**What do you see now?**

Minimum current consumption of 33uA, normal current consumption during join / send procedure.

**What do you want to see instead?**

A minimum current consumption of <5uA.

Has anybody experienced similar behaviour? Anybody able to have lower current consumption in low power mode? 
 @azerimaker

azerimaker, 2023-04-12:
Hi, @ThomasBaumgartner 

thank you for raising this issue. The extra ~31uA is usually the result of the battery monitor circuit being left ON. This was a minor bug which was addressed in the `sensors_lorawan_se` app. Please refer to [Software/app/sensors_lorawan_se/sensors.c](https://github.com/TheThingsIndustries/generic-node-se/commit/532066aa9d1b59a6e9d9434ae63ec6c2640ed7a8) to make sure your code follows the same sequence and let us know the results.
I hope it helps.

closed 2023-04-12 by ThomasBaumgartner

azerimaker, 2023-04-13:
Happy to help, @ThomasBaumgartner .