TheThingsIndustries/generic-node-se #249: Generic Node SE Edition on AU915 band

ingmanuelalfonso, 2023-04-17:
**Summary:** 
I am testing for the first time the Generic Node SE and found that using the provided libraries something is wrong when tryed the AU915 band. The device came with EU868 but its possible to enable US915 but not AU915. Some had this problem?. 

...

**Steps to Reproduce:** 

1. ... Using STM32CUBE IDE 
2. ... On lora_app.h edited ACTIVE_REGION to LORAMAC_REGION_AU915
3. ... In lora_info.c the REGION_AU915 is in gray color as other bands.

I noted that is possible to disolder hardware components that goes to Ceramic Antena and solder ones to go to the external antenna connector but sure this is not the problem

ingmanuelalfonso, 2024-05-20:
I came back to the same topic to inform I am able to communicate using AU915. 

- I am using sensors_lorawan App example
- I am using Chirpstack 3.0 as a network and application server
- I am using OTAA.
- I am using RAk7271 8 channel gateway connected to a raspberry pi 3b+

closed 2024-05-20 by ingmanuelalfonso

azerimaker, 2024-05-21:
Thank you for the update, @ingmanuelalfonso.